LoRaMesher/LoRaMesher #6: TODO List

Jaimi5, 2022-06-13:
### Overview 
In this page, we will comment what should be done to accomplish a fully functional library.

Feel free to comment for new ideas!

### Features
- [ ] Test timeout for routing tables 
- [ ] Add timeout for the large and reliable payloads
     - [ ] Consider how many hops to reach a node
     - [ ] Consider duty cycle
     - [ ] Consider traffic delays
     - [ ] Test timeouts

- [ ] Improve duty cycle, do it with the number of bytes sent
- [ ] Pause LoRaMesher

- [ ] FreeRTOS
     - [ ] Use Queues from FreeRTOS
     - [ ] Use memory allocation from FreeRTOS

- [ ] When sending a message, assign an id and return to user. When send notice the user that has been send.

### Future ideas
- [ ] A real application example using LoRaMesher

vinieich, 2023-02-09:
Hello
This is a very interesting library, nice work. 
I am engineer student and I'm trying to use this library with some heltec V2 LoRa.
I'm only doing the firsts tests now but there is something that I would like to ask if it's not any problem.

Is there any way to recover from esp32 deep sleep and send a packet without the routing table? Like if there is a node that is assured to be on and in range. Or maybe is there a way to export the routing table? So then it can be saved in the RTC memory that is saved on deep sleep wake up.

Thanks in advance and sorry for any grammar mistakes, english is not my primary language.

EddieCarrera, 2023-10-12:
Do you think trying to implement the STM32WL would be practical? It should have enough flash (256KB), but the amount of RAM is limiting (64KB). I'm coming from Meshtastic and their implementation is already too big for the 256KB limitation. 

I did some initial changes to convert to [STM32FreeRTOS](https://github.com/stm32duino/STM32FreeRTOS), but refactoring the code to remove the ESP32 and RadioLib module object dependencies, to STM32 and RadioLab's STM32WL object seems like it would take some effort.

Just wanted to ask if you think the hardware specs would be too restraining for LoRaMesher before I try to pursue it.