LoRaMesher/LoRaMesher #10: Doubt: a node receives a data message but the best route to the destination is the node it received it from

rogerpueyo, 2022-08-13:
Hi, @Jaimi5,

Sorry to bother you, I'd like to ask you a quick question about the routing protocol implementation. How do you avoid loops?

**Context**
Imagine you've got nodes A and B as part of a greater mesh network with other nodes. Node S, somewhere in the network, is sending a message to node R, which is far away. Nodes A and B are in the middle of the path between S and R.

Node A receives the data message mentioned above. The _next hop_ field in the message is node A, so it now has to forward it towards node R. According to the routing table at A, the best hop to reach R is via B, so node A modifies the _next hop_ accordingly and forwards the data message to B.

**Issue**
Node B receives the message from B headed towards R. However, for some reason (the node rebooted, the network is overloaded, routing table corruption... whatever), node B's routing table indicates that the best route towards R is via A.

If now node B forwards the message to A, they'll most likely be creating a loop.

**Possible solutions**
The issue above is a well-known problem in distance-vector routing protocols, for which different complementary solutions exist:
- Upon receiving a message (in this case, at node B), check if the next hop in the routing table towards R is different to the node where the message arrived from; if they're the same, delete the message
- Upon receiving a message (in this case, at node A), check whether that message had been already received from another path (if so, delete it)
- Add a TTL/max_hop field to the data message, so that the message expires after a certain number of hops

Does LoRaMesher implement any of the above or other strategies to avoid/mitigate loops?

Thanks!

rogerpueyo, 2022-08-14:
Thinking further the above, another possible approach would be to add a field to the packet header specifying the path metric or the path cost. If node B receives the packet mentioned above from node A with cost _n_, and the route towards node R has a higher cost than _n_, this means something wrong is happening there. I don't know if this is enough to avoid loops...

Also, loops can be not only node A => node B => node A => node B, but can involve more nodes.

Jaimi5, 2022-08-16:
Hi @rogerpueyo!

At this time, the only metric that is shared is the number of hops to reach a node. And only the best route is stored. For this reason, the nodes that have weak links are not allways reachables.
I am working now with another algorithm to send the snr between nodes and each one would choose the best route to a next hope. Moreover, in the future I would add more parammeters to the path cost. However, the loops in this algorithm could happend and the idea that you mention to share this metric inside the packet header would solve the issue.

In the next month, I hope I could finish this algorithm and test it in a real life network.

Do you have more ideas to improve the algorithm?

Thanks for the comment!

rogerpueyo, 2022-08-22:
Hi, @Jaimi5,

Thanks for your reply. Let me please answer in-line. :-)

> At this time, the only metric that is shared is the number of hops to reach a node. And only the best route is stored. For this reason, the nodes that have weak links are not allways reachables.

Nice, the hop-count metric is the safe one to start with.

> I am working now with another algorithm to send the snr between nodes and each one would choose the best route to a next hope. Moreover, in the future I would add more parammeters to the path cost.

Excellent. Number of hops is the simplest, but not always the best metric.

> However, the loops in this algorithm could happend and the idea that you mention to share this metric inside the packet header would solve the issue.

Loops are a difficult problem to solve, and there's plenty of literature regarding the topic. However, in your environment (i.e., scarce computational and airtime/bandwidth resources), I believe only few options will be available that are not overkill.

A simple strategy is to add an expiration timer to routes, and propagate routes between nodes _including_ this timer. This way, when a node disappears from the network, after some time, the rest of members will forget about it. You "just" need to properly set this expiration timer, to make sure routes are not incorrectly deleted when a node with bad connection still is available, which is non-trivial. I _believe_ this is already in your code, though.

While this does _not_ avoid loops, I believe adding a TTL (i.e., a counter which is decreased on every hop) field to the each LoRaMesher packet header can help mitigating their effect. This ensures that, if a packet enters a loop, it is deleted sooner or later.

> In the next month, I hope I could finish this algorithm and test it in a real life network.
 
If you find it convenient, please let me know about any progress.

> Do you have more ideas to improve the algorithm?

If you want, we can discuss it (in person or via videoconference) in a couple of weeks. 

Thanks!