LoRaMesher/LoRaMesher #12: Support for SX1262

JJJS777, 2023-02-10:
Hello, 

I am trying to implement a project with LoRaMesher. My esp32 has a SX1262 LoRa chip. I saw that you have tagged the SX126x in the project, but I can't find any implementation with it. Do I have to adapt the LoRaMesher to the SX1262 myself?

I have already adapted the counter example and get it to run, unfortunately only the reception of packets does not work yet.

Do you have vllt a tip for me? 
Thanks for your help! 

Translated with www.DeepL.com/Translator (free version)

Jaimi5, 2023-02-13:
Hi!

It is not yet implemented the support for SX126X but it should be preaty easy to implement it!
I will start and I will tell you when it is finished!

JJJS777, 2023-02-15:
thank you for your reply. I have done everything so far only the receiving of packets does not work yet. do you think it could have something to do with 
```cpp
void LoraMesher::setDioActionsForScanChannel()
{
    // set the function that will be called
    // when LoRa preamble is not detected within CAD timeout period
    // or when a packet is received
    radio->setDio1Action(onReceivingTimeout);

    // set the function that will be called
    // when LoRa preamble is detected
    radio->setDio1Action(onReceive);
}

void LoraMesher::setDioActionsForReceivePacket()
{
    radio->setDio1Action(onReceive);
    radio->setDio1Action(onReceivingTimeout);
}

void LoraMesher::clearDioActions()
{
    Log.verboseln(F("Clearing callback function"));
    radio->clearDio1Action();
}
```
in [LoraMesher.cpp](https://github.com/LoRaMesher/LoRaMesher/blob/d9d923679e97ad58d79bc5f76ada02ad5e8f9ba0/src/LoraMesher.cpp#L107)? 
I changed all three ```setDio0Action```to ```setDio1Action``` because RadioLib only provides a [setDio1Action](https://jgromes.github.io/RadioLib/class_s_x126x.html#a0da667fe702d7b4aafaa4bf7e69ea40d) method for the sx1262

Jaimi5, 2023-02-16:
Hi!

This would not work correctly. You could remove the setDioActionsForScanChannel, and remove the onReceivingTimeout.
Then in the send routine you would need to remove the startscanchannel and probably this would work.

If you require further assistance, please contact me through my email, and we will try to arrange a meeting.

Jaimi5, 2023-03-20:
This issue has been fixed with the following pull request (#14). 
Anything else reopen the issue!

Thanks!

closed 2023-03-20 by Jaimi5