LoRaMesher/LoRaMesher #15: port to nRF52840

SYSTEL-NLP, 2023-03-23:
Hello,

Thanks a lot for your work. It'e a great solution.
I plan to port code to Nordic nRF52840, but I don't knowsn if it is possible according to need mémory and mcu power.
What do you think about it?

By advance, thank for your reponse.

Jaimi5, 2023-03-24:
Hey there!
To run LoRaMesher, you'll need approximately 25KB of flash and 400KB of RAM, but this may vary depending on the size of the Routing table and other parameters. I haven't tested it with a 64MHz CPU yet, so I'm not sure if it will impact packet reception since I'm currently testing it on a 240MHz CPU.

If I can help with anything else just let me know!

SYSTEL-NLP, 2023-03-24:
Hi,
Thanks for your response.
I tried to compil with Arduino PBS and board RAK4631 of RAKWireless (nRF52840, SX1262)
URL Board :[ https://raw.githubusercontent.com/RAKwireless/RAKwireless-Arduino-BSP-Index/main/package_rakwireless_index.json](url)
But without succes, I seem to have a problem with RadioLib.
 You write you use a mod RadioLib version.
Your help is weelcom :-)

`In file included ...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/freertos/portable/GCC/nrf52/portmacro.h:33,
                 from ...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/freertos/Source/include/portable.h:52,
                 from ...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/freertos/Source/include/FreeRTOS.h:62,
                 from ...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/rtos.h:47,
                 from ...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/Arduino.h:48,
                 from customcode\libraries\RadioLib\src/BuildOpt.h:6,
                 from customcode\libraries\RadioLib\src/TypeDef.h:4,
                 from customcode\libraries\RadioLib\src/RadioLib.h:38,
                 from customcode.h:5,
                 from customcode.cpp:1:
customcode.cpp: In static member function 'static void LoraCode::onReceive()':
...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/freertos/portable/CMSIS/nrf52/portmacro_cmsis.h:101:25: error: expected primary-expression before '!=' token
  101 |     if( xSwitchRequired != pdFALSE ) {                \
      |                         ^~
...\Arduino15\packages\rakwireless\hardware\nrf52\1.3.3\cores\nRF5/freertos/portable/CMSIS/nrf52/portmacro_cmsis.h:108:33: note: in expansion of macro 'portEND_SWITCHING_ISR'
  108 | #define portYIELD_FROM_ISR( x ) portEND_SWITCHING_ISR( x )
      |                                 ^~~~~~~~~~~~~~~~~~~~~
...customcode.cpp:224:9: note: in expansion of macro 'portYIELD_FROM_ISR'
  224 |         portYIELD_FROM_ISR();
      |         ^~~~~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1`

Jaimi5, 2023-03-24:
Can you share with me the code you're currently using? It seems like you're using a code that utilizes RadioLib, but I'm having trouble identifying it at the moment. By sharing the code, I can better assist you in understanding and troubleshooting any issues you may be encountering.

SYSTEL-NLP, 2023-03-25:
Of course, https://github.com/SYSTEL-NLP/LoRaM.git
I have invited you to this repository
I just adapted file path for "include"
I compil with Arduino IDE and board RAK4631 of RAKWireless (nRF52840, SX1262)
URL Board :[ https://raw.githubusercontent.com/RAKwireless/RAKwireless-Arduino-BSP-Index/main/package_rakwireless_index.json](https://github.com/LoRaMesher/LoRaMesher/issues/url)

Jaimi5, 2023-03-25:
Have you tried a simple example using FreeRTOS and RadioLib?

Jaimi5, 2023-04-03:
Could you try this new branch?
https://github.com/LoRaMesher/LoRaMesher/tree/RemovedWiFi
I removed the wifi dependence, however, I added an esp32 function, hope it works for you.