LoRaMesher/LoRaMesher #16: Support for HopeRF rfm95w

ahmadm08, 2023-04-01:
hi, @Jaimi5 

Thanks a lot for your work. It's a great project.
i'm an electrical engineering student. i consider to use this for the monitoring system on my final year project.

will there be support for rfm95w,
or what should i change on the code to add support for the module?

Jaimi5, 2023-04-03:
Hi @ahmadm08
I implemented the support for rfm95, you can try it here: [RFM95](https://github.com/LoRaMesher/LoRaMesher/tree/module/RFM95)
You would need to change the LoRaChips acordingly and when initializing LoRaMesher remember to put the first parameter to LM_RFM95 to initialize the correct module.

Hope it helps!
If it works correctly tell me and I would add it into the LoRaMesher.
Thanks!

Jaimi5, 2026-10-08:
RFM95W is SX1276-based: use `RadioType::kSx1276` in 2.0.0 (#130). Will close on merge.