LoRaMesher/LoRaMesher #18: Larger message support

blackcoffeexbt, 2023-07-14:
In the GitHub.io page for the project there is mention of support for larger messages 

"Protocol
...
Send large payloads. Since we cannot send packets larger than 222 bytes, we divide the payload and send reliable messages until is all sent/received"

Could you please point me in the direction of how I can use this? I've tested with larger AppPackets but it isn't working correctly, I've also read the example code but I'm not sure if it's completed. 

Is this large payload support active?

Thanks

Jaimi5, 2023-07-17:
Hi!!
Currently we are working on it but do not yet have a stable version. Soon I will have a tested and I will publish it officially!

Are you interested to use LoRaMesher? Maybe you can tell us more about your application. Our implementation for sending large payloads is closely related to the features of LoRaMesher.

blackcoffeexbt, 2023-07-17:
Thanks for the reply @Jaimi5 

I'm very interested in LoRaMesher. I'm building a mesh network that will use LoRa devices to send "notes" from mesh network devices to the internet, broadcasting from nodes with an internet connection to Nostr relays.

Nostr notes are 350 bytes minimum so being able to send larger packets with LoRaMesher would be extremely useful.

Thanks

Jaimi5, 2023-10-05:
Hi again!

A new version has been updated. It seems to work preaty well. I need to document it yet but you have an example on how to send a large and reliable payload to a node.
You can set nodes roles, for example a gateway role, and it will retransmited to all the nodes with the routing table.
Additionally you can get the nearest gateway.

I have an implementation here if you want to see how to use everything [here](https://github.com/Jaimi5/LoRaChat/blob/espidfwifiAndMqtt/src/loramesh/loraMeshService.cpp). folder 

Hope this helps!

closed 2023-10-05 by Jaimi5