LoRaMesher/LoRaMesher #21: Add SD card logging

PureTek-Innovations, 2023-08-07:
Hi @Jaimi5 
I'm having trouble with nodes locking up or stopping forwarding messages after what seems to be a random amount of time.
I'm currently using the simulator branch but had the same or similar issues with the main branch.
Sometimes they crash, sometimes they continue to send their messages but stop forwarding other messages.
Most of my installs are hard to physically access, but I have tried connecting a USB cable to a crashed node, but there is no serial output.
I'm using TTGO V2.1 modules with an SD card slot, is it possible to add logging to the SD card?
I've looked at the logging library and it's not obvious to me.
If I can log to the SD card then I have half a chance of tracing the reason.
Thank you very much

sergimn, 2023-08-08:
This should be done in the logging library level ([ArduinoLog](https://github.com/thijse/Arduino-Log) as of today) in my opinion.

Regarding the hanging of nodes, this is most likely to either a memory leak or the WDT not being reset on time, but you could also try to check the [reason why the device got reset](https://github.com/espressif/arduino-esp32/issues/449)

Jaimi5, 2023-08-08:
Hi @PureTek-Innovations! Since we are using [ArduinoLog](https://github.com/thijse/Arduino-Log) as mentioned by @sergimn, you can add the following code after initializating LoRaMesher.

```
    #include <SD.h>
    File myFile;

    // Create/Open file 
    myFile = SD.open("test.txt", FILE_WRITE);

    if (myFile) {
        Log.begin(LOG_LEVEL_VERBOSE, &myFile);
    }

```

Next week I will try to look if there is a memory leak, however, one of those has been solved here: e1f5fe3cea94afdfddf8b65d5387c89996f64748

Hope it helps!

PureTek-Innovations, 2023-08-10:
Thanks @Jaimi5 
The code snippet is not complete.

I have tried this:
`#define SDCARD_MOSI 15
#define SDCARD_MISO 2
#define SDCARD_SCLK 14
#define SDCARD_CS   13
File logFile;

void setupSDLogging(){
    if (SDCARD_CS >  0) {
        SPIClass sdSPI(VSPI);
        sdSPI.begin(SDCARD_SCLK, SDCARD_MISO, SDCARD_MOSI, SDCARD_CS);
        if (!SD.begin(SDCARD_CS, sdSPI)) {
            Serial.println( "SDCard  FAIL");
        } else {
            Serial.println("SDCard  PASS");
            uint32_t cardSize = SD.cardSize() / (1024 * 1024);
            Serial.print( "Size: ");
            Serial.print(String(cardSize));
            Serial.println(" MB");
        }
        delay(2000);
    }

    // Create/Open file 
    logFile = SD.open("/log.txt", FILE_WRITE);

    if (logFile) {
        // Log.begin(LOG_LEVEL_VERBOSE, &logFile);
    }
}`
The card mounts and I get the size printed to the serial monitor.
But then it crashes with this error:

`V: Lora initialized
[   862][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x400dffc4 arg=0x3ffbdc64
SDCard  PASS
Size: 30528 MB
[  3376][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  3376][E][sd_diskio.cpp:621] ff_sd_status(): Check status failed

assert failed: xQueueSemaphoreTake queue.c:1549 (pxQueue->uxItemSize == 0)

Backtrace: 0x40083c61:0x3ffb1a20 0x4008d32d:0x3ffb1a40 0x4009332d:0x3ffb1a60 0x4008e33d:0x3ffb1b90 0x400d446c:0x3ffb1bd0 0x400dbc3b:0x3ffb1c00 0x400dc207:0x3ffb1c30 0x4010b8e5:0x3ffb1c70 0x4010bae4:0x3ffb1c90 0x4010bb1d:0x3ffb1cb0 0x4010bc3a:0x3ffb1cd0 0x4010db63:0x3ffb1d10 0x4010fe87:0x3ffb1f80 0x400ed4e6:0x3ffb20f0 0x4015fa81:0x3ffb2110 0x40154d23:0x3ffb2130 0x400dba91:0x3ffb21c0 0x400d38c9:0x3ffb21f0 0x400d3df1:0x3ffb2260 0x400e08e2:0x3ffb2290

  #0  0x40083c61:0x3ffb1a20 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008d32d:0x3ffb1a40 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x4009332d:0x3ffb1a60 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x4008e33d:0x3ffb1b90 in xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:1549 (discriminator 1)
  #4  0x400d446c:0x3ffb1bd0 in SPIClass::beginTransaction(SPISettings) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:178
  #5  0x400dbc3b:0x3ffb1c00 in (anonymous namespace)::AcquireSPI::AcquireSPI(ardu_sdcard_t*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/sd_diskio.cpp:468
  #6  0x400dc207:0x3ffb1c30 in ff_sd_read(unsigned char, unsigned char*, unsigned int, unsigned int) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/SD/src/sd_diskio.cpp:635
  #7  0x4010b8e5:0x3ffb1c70 in ff_disk_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/diskio/diskio.c:70      
  #8  0x4010bae4:0x3ffb1c90 in move_window at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c:1077
      (inlined by) move_window at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c:1064
  #9  0x4010bb1d:0x3ffb1cb0 in check_fs at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c:3193
  #10 0x4010bc3a:0x3ffb1cd0 in find_volume at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c:3272
  #11 0x4010db63:0x3ffb1d10 in f_stat at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c:4610
  #12 0x4010fe87:0x3ffb1f80 in vfs_fat_stat at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/vfs/vfs_fat.c:597       
  #13 0x400ed4e6:0x3ffb20f0 in esp_vfs_stat at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/vfs/vfs.c:580 (discriminator 3)
  #14 0x4015fa81:0x3ffb2110 in stat at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/syscalls/sysstat.c:11
  #15 0x40154d23:0x3ffb2130 in VFSImpl::open(char const*, char const*, bool) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.cpp:44
  #16 0x400dba91:0x3ffb21c0 in fs::FS::open(char const*, char const*, bool) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.cpp:234
  #17 0x400d38c9:0x3ffb21f0 in setupSDLogging() at src/main.cpp:332
  #18 0x400d3df1:0x3ffb2260 in setup() at src/main.cpp:353
  #19 0x400e08e2:0x3ffb2290 in loopTask(void*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42   

ELF file SHA256: e81d44c465c1def3`

PureTek-Innovations, 2023-08-11:
I just tried HSPI instead of VSPI
No boot loops, but now the card isn't mounting

`SPIClass sdSPI(HSPI);`

`10:18:31.446 > : Lora initialized
10:18:31.446 > [  1361][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
10:18:31.446 > [  1861][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
10:18:31.446 > SDCard  FAIL
10:18:31.446 > [  3862][E][vfs_api.cpp:24] open(): File system is not mounted`

PureTek-Innovations, 2023-08-12:
I've managed to get the SD Card to mount properly with the SD_MMC library using this example
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino
I had to change lib_ldf_mode to deep rather than deep+
I had to add 
pinMode(SDCARD_MISO, INPUT_PULLUP);
The tests included with the example all work properly.

There are no errors, but I cannot get the logging library to actually write any data to the SD card.

PureTek-Innovations, 2023-08-16:
To be honest, I'm not convinced that the logging library actually supports writing logs to an SD card.
This one looks promising https://github.com/embeddedartistry/arduino-logger
But it seems to be optimised for Teensy MCUs.  
There is an outstanding pull request to make it work with ESP8266s which may also work with ESP32s.
https://github.com/embeddedartistry/arduino-logger/pull/18

Jaimi5, 2023-08-16:
Hi again!
I am trying to find the bug however I cannot help you with the SD card problem because I don't have any SD card module...
Maybe you could change the logging library to print it directly to the SD card without initializing it.
I am testing the library for more than three days straight and still not having the bug... hope it occurs soon and I locate it.
I will let you inform.

PureTek-Innovations, 2023-08-16:
Hi
I've managed to get the network to be reasonably stable.
I set the two worst nodes to reboot every 4 hours - and now it seems to work.

My best guess is that there is a routing issue, I've had this before.
The best route is 
Node 4 to Node 3 to Node 2 to Node 1
Sometimes there is a weak signal from Node 4 to Node 2, and it adds that route, but the route doesn't actually work.
I've set nodes 4 and 2 to reboot every 4 hours.
Node 3 is very hard to access, it's mounted very high on a tower.

It's been stable for a couple of days now.

Jaimi5, 2023-08-16:
That's kind of good! Thanks for informing!
With other projects I was able to have a good connection putting a higher spreading factor and less MAX PACKET SIZE. Maybe it helps for you!

Do you know how big are the packets?

I will still search for this bug!
Thanks for the patience!

PureTek-Innovations, 2023-08-16:
The packets are 12 bytes
Maximum packet size is set to 222 bytes

The radio settings are:
#define LM_BAND 863.0F
#define LM_BANDWIDTH 500.0F
#define LM_LORASF 7U // Spreading factor 6-12 (default 7)
#define LM_CODING_RATE 7U
#define LM_PREAMBLE_LENGTH 8U
#define LM_POWER 10

PureTek-Innovations, 2023-08-16:
They are 100 miles away.
Next time I'm onsite, I'll only be there to take the network down.
I can try these settings at the next install.

Jaimi5, 2023-08-16:
Yes, exactly! That's something that I am working on! It's important to synchronize all devices when changing these settings to avoid breaking the network.

Jaimi5, 2023-10-05:
Hi!
Now the Library uses ESP_LOG, which allows directly to specify the Serial on which do you want to print the data.
Hope it is better now!

Thanks!

closed 2023-10-05 by Jaimi5