LoRaMesher/LoRaMesher #30: Potential STM32WL Support

EddieCarrera, 2023-10-13:
Do you think trying to implement the STM32WL would be practical? It should have enough flash (256KB), but the amount of RAM is limiting (64KB). I'm coming from Meshtastic and their implementation is already too big for the 256KB limitation.

I did some initial changes to convert to [STM32FreeRTOS](https://github.com/stm32duino/STM32FreeRTOS), but refactoring the code to remove the ESP32 and RadioLib module object dependencies, to STM32 and RadioLab's STM32WL object seems like it would take some effort.

Just wanted to ask if you think the hardware specs would be too restraining for LoRaMesher before I try to pursue it.

Jaimi5, 2023-12-07:
Hi @EddieCarrera!
Based on my experience, the current implementation typically requires around 150KB, but this may vary depending on the network traffic.
We are indeed considering plans to evolve LoRaMesher into a more versatile solution, independent of freeRTOS. However, due to current time constraints, I may not be able to undertake this task in the upcoming months.

I will keep you updated and let you know as soon as I begin testing the new implementation.

I appreciate your understanding and patience.

Jaimi5, 2026-10-08:
STM32 work in progress: #122.