LoRaMesher/LoRaMesher #32: Compatibility with LoraWAN gateway and LoRaWAN concentrator

aluz123, 2023-11-23:
Wondering if there's any compatibility with loraWAN gateways that use LoRaWAN concentrators. Specifically the Draguino DLOS8 that uses a  SX1301 LoRaWAN concentrator. 1 x SX1301 + 2 x 1257 LoRa Tranceiver 

https://www.dragino.com/products/lora-lorawan-gateway/item/160-dlos8.html

The DLOS8 uses semtech packet forwarder and fully compatible with LoRaWAN protocol. It includes a SX1301 LoRaWAN concentrator, which provide 10 programmable parallel demodulation paths.

DLOS8 has pre-configured standard LoRaWAN frequency bands to use for different countries. User can also customized the frequency bands to use in their own LoRaWAN network.

DLOS8 can communicate with ABP LoRaWAN end node without LoRaWAN server. System integrator can use it to integrate with their existing IoT Service without set up own LoRaWAN server or use 3rd party LoRaWAN service.

PureTek-Innovations, 2023-11-30:
I don't maintain this library, but I do use it.
The underlying radio library, https://github.com/jgromes/RadioLib, does not support SX1301s
And I'm pretty sure that the hardware in the DLOS8 does not support [FreeRTOS](https://freertos.org/index.html).
This library needs both of these, see the dependencies section of the readme

aluz123, 2023-11-30:
@PureTek-Innovations I appreciate the response, thank you. Do you by chance know of any Lora mesh libraries that work with gateways? I'm looking to develop a solution that uses nodes and gateways as a mesh-based remote sensor monitoring system.