LoRaMesher/LoRaMesher #39: Watchdog/Crash after Error -705 in startReceiving()

samsondw, 2023-12-15:
I have been testing LoRaMesher with two Heltec Wireless Tracker devices (ESP32-S3 with SX1262).  I have a task that broadcasts a message every 5 seconds.  While things run okay when the units first start up, the communications seem to stop when left to run for extended periods. It seems there is something that causes the SPI interface to timeout when trying to receive, and the library is unable to recover.  

This is the most common sequence observed:
- Starting receiving gave error: -705 (once or multiple times)
- Either Watchdog triggered or Core panic follows.

#Watchdog Crash Log
```
13:20:56.098 > [2131410][V][LoraMesher.h:240] createPacketAndSend(): [LoRaMesher] Creating a packet for send with 6 bytes
13:20:56.106 > [2131416][V][PacketService.h:312] createPacket(): [LoRaMesher] Creating packet with 15 bytes
13:20:56.114 > [2131424][I][PacketService.h:325] createPacket(): [LoRaMesher] Packet created with 15 bytes
13:20:56.121 > [2131432][I][LoraMesher.h:648] setPackedForSend(): [LoRaMesher] Adding packet to Q_SP
13:20:56.129 > [2131440][I][LoraMesher.h:650] setPackedForSend(): [LoRaMesher] Created packet to Q_SP
13:20:56.140 > [2131447][I][LoraMesher.cpp:951] addToSendOrderedAndNotify(): [LoRaMesher] Added packet to Q_SP, notifying sender task
13:20:56.432 > [2131750][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
13:20:57.432 > [2132750][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
13:20:58.136 > E (2137118) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
13:20:58.137 > E (2137118) task_wdt:  - IDLE (CPU 0)
13:20:58.141 > E (2137118) task_wdt: Tasks currently running:
13:20:58.145 > E (2137118) task_wdt: CPU 0: Sending routine
13:20:58.148 > E (2137118) task_wdt: CPU 1: IDLE
13:20:58.151 > E (2137118) task_wdt: Aborting.
13:20:58.510 > 
13:20:59.679 > abort() was called at PC 0x42028690 on core 0
13:20:59.679 > 
13:20:59.679 > 
13:20:59.679 > Backtrace: 0x40377bee:0x3fc99290 0x4037fee1:0x3fc992b0 0x4038643d:0x3fc992d0 0x42028690:0x3fc99350 0x40378e21:0x3fc99370 0x42051aa3:0x3fca9810 0x4204d308:0x3fca9830 0x4204d680:0x3fca9860 0x4204d6ad:0x3fca98a0 0x4204de65:0x3fca98e0 0x4204f1cc:0x3fca9910 0x4204f24f:0x3fca9940 0x4204da0f:0x3fca9960 0x42069ac9:0x3fca9980 0x420060d0:0x3fca99a0 0x4200611a:0x3fca99e0 0x4200611a:0x3fca9a20 0x420069d7:0x3fca9a60 0x42008b50:0x3fca9ab0 0x42008c25:0x3fca9b00
13:20:59.679 > 
13:20:59.679 >   #0  0x40377bee:0x3fc99290 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
13:20:59.679 >   #1  0x4037fee1:0x3fc992b0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
13:20:59.679 >   #2  0x4038643d:0x3fc992d0 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
13:20:59.679 >   #3  0x42028690:0x3fc99350 in task_wdt_isr at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/task_wdt.c:176 (discriminator 3)
13:20:59.679 >   #4  0x40378e21:0x3fc99370 in _xt_lowint1 at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_vectors.S:1114
13:20:59.679 >   #5  0x42051aa3:0x3fca9810 in ArduinoHal::yield() at lib/RadioLib/src/ArduinoHal.cpp:237
13:20:59.679 >   #6  0x4204d308:0x3fca9830 in Module::SPItransferStream(unsigned char*, unsigned char, bool, unsigned char*, unsigned char*, unsigned int, bool, unsigned int) at lib/RadioLib/src/Module.cpp:292
13:20:59.679 >   #7  0x4204d680:0x3fca9860 in Module::SPIwriteStream(unsigned char*, unsigned char, unsigned char*, unsigned int, bool, bool) at lib/RadioLib/src/Module.cpp:229
13:20:59.679 >   #8  0x4204d6ad:0x3fca98a0 in Module::SPIwriteStream(unsigned char, unsigned char*, unsigned int, bool, bool) at lib/RadioLib/src/Module.cpp:224
13:20:59.679 >   #9  0x4204de65:0x3fca98e0 in SX126x::setDioIrqParams(unsigned short, unsigned short, unsigned short, unsigned short) at lib/RadioLib/src/modules/SX126x/SX126x.cpp:1819
13:20:59.679 >   #10 0x4204f1cc:0x3fca9910 in SX126x::startReceiveCommon(unsigned int, unsigned short, unsigned short) at lib/RadioLib/src/modules/SX126x/SX126x.cpp:676
13:20:59.679 >   #11 0x4204f24f:0x3fca9940 in SX126x::startReceive(unsigned int, unsigned short, unsigned short, unsigned int) at lib/RadioLib/src/modules/SX126x/SX126x.cpp:594
13:20:59.679 >   #12 0x4204da0f:0x3fca9960 in SX126x::startReceive() at lib/RadioLib/src/modules/SX126x/SX126x.cpp:589
13:20:59.679 >   #13 0x42069ac9:0x3fca9980 in LM_SX1262::startReceive() at lib/LoRaMesher/src/modules/LM_SX1262.cpp:16
13:20:59.679 >   #14 0x420060d0:0x3fca99a0 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:179
13:20:59.679 >   #15 0x4200611a:0x3fca99e0 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
13:20:59.679 >   #16 0x4200611a:0x3fca9a20 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
13:20:59.679 >   #17 0x420069d7:0x3fca9a60 in LoraMesher::sendPacket(Packet<unsigned char>*) at lib/LoRaMesher/src/LoraMesher.cpp:414 (discriminator 2)
13:20:59.679 >   #18 0x42008b50:0x3fca9ab0 in LoraMesher::sendPackets() at lib/LoRaMesher/src/LoraMesher.cpp:476
13:20:59.679 >   #19 0x42008c25:0x3fca9b00 in LoraMesher::initializeSchedulers()::{lambda(void*)#2}::_FUN(void*) at lib/LoRaMesher/src/LoraMesher.cpp:225
13:20:59.679 >       (inlined by) _FUN at lib/LoRaMesher/src/LoraMesher.cpp:225
```

#Core Panic Crash Log
```
19:55:23.323 > [3387552][V][LoraMesher.h:240] createPacketAndSend(): [LoRaMesher] Creating a packet for send with 6 bytes
19:55:23.331 > [3387558][V][PacketService.h:312] createPacket(): [LoRaMesher] Creating packet with 15 bytes
19:55:23.340 > [3387566][I][PacketService.h:325] createPacket(): [LoRaMesher] Packet created with 15 bytes
19:55:23.348 > [3387574][I][LoraMesher.h:648] setPackedForSend(): [LoRaMesher] Adding packet to Q_SP
19:55:23.355 > [3387582][I][LoraMesher.h:650] setPackedForSend(): [LoRaMesher] Created packet to Q_SP
19:55:23.364 > [3387590][I][LoraMesher.cpp:951] addToSendOrderedAndNotify(): [LoRaMesher] Added packet to Q_SP, notifying sender task
19:55:24.126 > [3388361][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
19:55:25.127 > [3389361][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
19:55:26.126 > [3390361][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
19:55:27.127 > [3391361][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
19:55:28.126 > [3392361][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -705
19:55:28.371 > [3392600][V][LoraMesher.h:240] createPacketAndSend(): [LoRaMesher] Creating a packet for send with 6 bytes
19:55:28.379 > [3392606][V][PacketService.h:312] createPacket(): [LoRaMesher] Creating packet with 15 bytes
19:55:28.388 > [3392614][I][PacketService.h:325] createPacket(): [LoRaMesher] Packet created with 15 bytes
19:55:28.396 > [3392622][I][LoraMesher.h:648] setPackedForSend(): [LoRaMesher] Adding packet to Q_SP
19:55:28.402 > [3392630][I][LoraMesher.h:650] setPackedForSend(): [LoRaMesher] Created packet to Q_SP
19:55:28.413 > [3392638][I][LoraMesher.cpp:951] addToSendOrderedAndNotify(): [LoRaMesher] Added packet to Q_SP, notifying sender task
[... 59 lines cut ...]
19:55:43.253 >   #22 0x4200611a:0x3fca9560 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #23 0x4200611a:0x3fca95a0 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #24 0x4200611a:0x3fca95e0 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #25 0x4200611a:0x3fca9620 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #26 0x4200611a:0x3fca9660 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #27 0x4200611a:0x3fca96a0 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #28 0x4200611a:0x3fca96e0 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
19:55:43.253 >   #29 0x4200611a:0x3fca9720 in LoraMesher::startReceiving() at lib/LoRaMesher/src/LoraMesher.cpp:182
[... cut ...]
19:55:43.253 >   #42 0x420069d7:0x3fca9a60 in LoraMesher::sendPacket(Packet<unsigned char>*) at lib/LoRaMesher/src/LoraMesher.cpp:414 (discriminator 2)
19:55:43.253 >   #43 0x42008b50:0x3fca9ab0 in LoraMesher::sendPackets() at lib/LoRaMesher/src/LoraMesher.cpp:476
19:55:43.253 >   #44 0x42008c25:0x3fca9b00 in LoraMesher::initializeSchedulers()::{lambda(void*)#2}::_FUN(void*) at lib/LoRaMesher/src/LoraMesher.cpp:225
19:55:43.253 >       (inlined by) _FUN at lib/LoRaMesher/src/LoraMesher.cpp:225
```

samsondw, 2023-12-15:
I modified startReceiving() to restart the radio before retrying.  I also modified sendPacket() as shown below.

```
int LoraMesher::startReceiving() {
    setDioActionsForReceivePacket();

    int res = radio->startReceive();
    if (res != 0) {
        ESP_LOGE(LM_TAG, "Starting receiving gave error: %d", res);
        if ((res == RADIOLIB_ERR_SPI_CMD_TIMEOUT) || (res == RADIOLIB_ERR_UNKNOWN)) {
            ESP_LOGW(LM_TAG, "Restarting radio");
            restartRadio();
        }
        res = startReceiving();
    }
    return res;
}

bool LoraMesher::sendPacket(Packet<uint8_t>* p) {
    waitBeforeSend(1);

    clearDioActions();

    // Print the packet to be sent
    printHeaderPacket(p, "send");

    //Blocking transmit, it is necessary due to deleting the packet after sending it.
    int resT = radio->transmit(reinterpret_cast<uint8_t*>(p), p->packetSize);
    bool ret = true;
    if (resT != RADIOLIB_ERR_NONE) {
        ESP_LOGE(LM_TAG, "Transmit gave error: %d", resT);
        ret = false;
    }

    //Start receiving again after sending a packet
    startReceiving();

    return ret;
}
```

Now it seems to recover gracefully without crashing:
```
14:06:11.060 > [3818949][I][LoraMesher.cpp:956] addToSendOrderedAndNotify(): [LoRaMesher] Added packet to Q_SP, notifying sender task
14:06:11.069 > [3818959][V][LoraMesher.cpp:444] sendPackets(): [LoRaMesher] Stack space unused after entering the task: 1944
14:06:11.076 > [3818969][V][LoraMesher.cpp:445] sendPackets(): [LoRaMesher] Free heap: 238824
14:06:11.084 > [3818976][V][LoraMesher.cpp:451] sendPackets(): [LoRaMesher] Size of Send Packets Queue: 1
14:06:11.091 > [3818984][V][LoraMesher.cpp:458] sendPackets(): [LoRaMesher] Send n. 784
14:06:11.098 > [3818990][V][LoraMesher.cpp:390] waitBeforeSend(): [LoRaMesher] RandomDelay 1057 ms
14:06:12.157 > [3820054][V][LoraMesher.cpp:689] printHeaderPacket(): [LoRaMesher] Packet send -- Size: 15 Src: A704 Dst: A708 Id: 16 Type: 2 Via: A708 Seq_Id: 0 Num: 0
14:06:12.164 > [3820056][E][LoraMesher.cpp:418] sendPacket(): [LoRaMesher] Transmit gave error: -707
14:06:12.173 > [3820064][E][LoraMesher.cpp:181] startReceiving(): [LoRaMesher] Starting receiving gave error: -1
14:06:12.180 > [3820072][W][LoraMesher.cpp:183] startReceiving(): [LoRaMesher] Restarting radio
14:06:12.193 > [3820092][V][LoraMesher.cpp:109] initializeLoRa(): [LoRaMesher] Initializing RadioLib
14:06:12.196 > [3820092][V][LoraMesher.cpp:124] initializeLoRa(): [LoRaMesher] Using SX1262 module
14:06:12.203 > [3820096][V][LoraMesher.cpp:146] initializeLoRa(): [LoRaMesher] Initializing radio
14:06:12.253 > [3820152][I][LoraMesher.cpp:155] initializeLoRa(): [LoRaMesher] LoRa module initialization DONE
14:06:12.258 > [3820153][I][LoraMesher.cpp:105] restartRadio(): [LoRaMesher] Restarting radio DONE
14:06:12.265 > [3820163][V][LoraMesher.cpp:451] sendPackets(): [LoRaMesher] Size of Send Packets Queue: 1
14:06:12.272 > [3820165][V][LoraMesher.cpp:458] sendPackets(): [LoRaMesher] Send n. 785
14:06:12.279 > [3820171][V][LoraMesher.cpp:390] waitBeforeSend(): [LoRaMesher] RandomDelay 1678 ms
14:06:13.959 > [3821857][V][LoraMesher.cpp:689] printHeaderPacket(): [LoRaMesher] Packet send -- Size: 15 Src: A704 Dst: A708 Id: 17 Type: 2 Via: A708 Seq_Id: 0 Num: 0
14:06:14.171 > [3822071][V][LoraMesher.cpp:508] sendPackets(): [LoRaMesher] TimeOnAir 197 ms, next message in 0 ms
```

samsondw, 2023-12-19:
UPDATE: After letting 2 units run over the weekend, I only logged one instance of a crash.  In this case there was a -707 error in startReceiving().  So it looks like it would be worth it to reset the radio after any error in startReceiving().

Jaimi5, 2023-12-19:
Thank you very much for the observations!
It seams like that the errors only occur with the SX126X module. 
I added that every time it occurs an error in the startReceiving it will reset the radio.
Let me know if this solution works!

Thanks again!

Jaimi5, 2024-01-21:
Since it has not been updated for one month, I consider it solved. 
Closing the issue. 
Thanks!

closed 2024-01-21 by Jaimi5

PureTek-Innovations, 2024-07-09:
This error is back, the reset code is no longer in the library.