LoRaMesher/LoRaMesher #40: Support for LoRa-CC68 LoRa module by NiceRF

Sir20PiR, 2024-01-03:
Thanks a lot for your work. It's a great project.

I am reaching out regarding the possibility of integrating support for the LLC68 chip within your repository. The LLC68 is a low-cost microcontroller from the STM32L0 series by STMicroelectronics. However, I've encountered challenges as most existing libraries do not currently support this specific chip.

Considering the potential benefits of including support for the LLC68 within your repository, I am eager to start a project utilizing this module later this year. If it's feasible, I kindly request your assistance in incorporating this chip. Alternatively, If it's not possible the integration would greatly appreciate any guidance or support in setting it up through custom configurations.

PureTek-Innovations, 2024-01-03:
Hi
The LLC68 appears to be a LoRa transceiver from Semtech.
https://www.semtech.com/products/wireless-rf/lora-connect/llcc68
The underlying radio library, radiolib, appears to support the LLC68.
https://github.com/jgromes/RadioLib/blob/master/keywords.txt

The STM32L0 is a microprocessor
https://www.st.com/en/microcontrollers-microprocessors/stm32l0-series.html

There is a fork of free RTOS for it:
https://github.com/stm32duino/STM32FreeRTOS

So it may be possible to run this library on an STM32L0 with an LLC68 transceiver depending on hardware resources and the like.

Watch out, though as there is also other ESP specific code in the library, like ESP_LOG.

Jaimi5, 2024-01-21:
Hi! Sorry for the delay on the response!
And thanks @PureTek-Innovations for answering!
What I have seen is that LLC68 is compatible directly with SX1262. Since it is implemented yet could you try to execute the examples initiating with this module?
I will work on how to add correctly the LLC68!

Thanks!

Sir20PiR, 2024-01-26:
@Jaimi5 
For sure! This weekend I will make some tests and I share in the thread.