LoRaMesher/LoRaMesher #43: Using t-beam-supreme-s3

COLVERTYETY, 2024-03-24:
Hello,
I am trying to use LoRaMesher with a t-beam-supreme-s3 and am encountering some difficulties.
I am trying to use the counter example.
The board uses sx1262. 
Here is the code I am using:

```
#include <Arduino.h>
#include "LoraMesher.h"
#include <SPI.h>

#define LoRa_frequency            915E6
#define RADIO_SCLK_PIN              (12)
#define RADIO_MISO_PIN              (13)
#define RADIO_MOSI_PIN              (11)
#define RADIO_CS_PIN                (10)
#define RADIO_DIO0_PIN               (-1)
#define RADIO_RST_PIN               (5)
#define RADIO_DIO1_PIN              (1)
#define RADIO_BUSY_PIN              (4)

LoraMesher& radio = LoraMesher::getInstance();

uint32_t dataCounter = 0;
uint32_t receiveCounter = 0;
struct dataPacket {
  uint32_t counter = 0;
};

dataPacket* helloPacket = new dataPacket;

/**
 * @brief Print the counter of the packet
 *
 * @param data
 */
void printPacket(dataPacket data) {
    Serial.printf("Hello Counter received nº %d\r\n", data.counter);
    receiveCounter++;
}

/**
 * @brief Iterate through the payload of the packet and print the counter of the packet
 *
 * @param packet
 */
void printDataPacket(AppPacket<dataPacket>* packet) {
    Serial.printf("Packet arrived from %X with size %d\n", packet->src, packet->payloadSize);

    //Get the payload to iterate through it
    dataPacket* dPacket = packet->payload;
    size_t payloadLength = packet->getPayloadLength();

    for (size_t i = 0; i < payloadLength; i++) {
        //Print the packet
        printPacket(dPacket[i]);
    }
}

/**
 * @brief Function that process the received packets
 *
 */
void processReceivedPackets(void*) {
    for (;;) {
        /* Wait for the notification of processReceivedPackets and enter blocking */
        ulTaskNotifyTake(pdPASS, portMAX_DELAY);

        //Iterate through all the packets inside the Received User Packets Queue
        while (radio.getReceivedQueueSize() > 0) {
            Serial.println("ReceivedUserData_TaskHandle notify received");
            Serial.printf("Queue receiveUserData size: %d\n", radio.getReceivedQueueSize());

            //Get the first element inside the Received User Packets Queue
            AppPacket<dataPacket>* packet = radio.getNextAppPacket<dataPacket>();

            //Print the data packet
            printDataPacket(packet);

            //Delete the packet when used. It is very important to call this function to release the memory of the packet.
            radio.deletePacket(packet);
        }
    }
}

TaskHandle_t receiveLoRaMessage_Handle = NULL;

/**
 * @brief Create a Receive Messages Task and add it to the LoRaMesher
 *
 */
void createReceiveMessages() {
    int res = xTaskCreate(
        processReceivedPackets,
        "Receive App Task",
        4096,
        (void*) 1,
        2,
        &receiveLoRaMessage_Handle);
    if (res != pdPASS) {
        Serial.printf("Error: Receive App Task creation gave error: %d\n", res);
    }

    radio.setReceiveAppDataTaskHandle(receiveLoRaMessage_Handle);
}

/**
 * @brief Initialize LoRaMesher
 *
 */
void setupLoraMesher() {
    // Example on how to change the module. See LoraMesherConfig to see all the configurable parameters.
    LoraMesher::LoraMesherConfig config;
    config.module = LoraMesher::LoraModules::SX1262_MOD;
    config.loraCs = RADIO_CS_PIN;
    config.loraIrq = RADIO_DIO0_PIN;
    config.loraRst = RADIO_RST_PIN;
    config.loraIo1 = RADIO_DIO1_PIN;
    config.freq = LoRa_frequency;
    //Init the loramesher with a processReceivedPackets function
    radio.begin(config);

    //Create the receive task and add it to the LoRaMesher
    createReceiveMessages();

    //Start LoRaMesher
    radio.start();

    Serial.println("Lora initialized");
}

void setup() {
    Serial.begin(115200);

    Serial.println("initBoard");
    setupLoraMesher();
}

void printRoutingTable() {

    //Set the routing table list that is being used and cannot be accessed (Remember to release use after usage)
    LM_LinkedList<RouteNode>* routingTableList = radio.routingTableListCopy();

    routingTableList->setInUse();

    Serial.println("Routing Table:");
    char text[15];
    for (int i = 0; i < radio.routingTableSize(); i++) {
        RouteNode* rNode = (*routingTableList)[i];
        NetworkNode node = rNode->networkNode;
        snprintf(text, 15, ("|%X(%d)->%X"), node.address, node.metric, rNode->via);
        Serial.println(text);
    }
    Serial.println("End Routing Table");

    //Release routing table list usage.
    routingTableList->releaseInUse();

    // Delete routing table list
    delete routingTableList;
}

void loop() {
    for (;;) {
        Serial.printf("Send packet %d\n", dataCounter);
        Serial.printf("receivedCounter: %d\n", receiveCounter);
        char addrStr[15];
        snprintf(addrStr, 15, "Id: %X\r\n", radio.getLocalAddress());
        Serial.println(addrStr);
        // print info
        printRoutingTable();

        helloPacket->counter = dataCounter++;

        //Create packet and send it.
        radio.createPacketAndSend(BROADCAST_ADDR, helloPacket, 1);

        //Wait 20 seconds to send the next packet
        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }
}

```

I am flashing this code on 2 separate devices.
The radio seams to init without issues.
However, The routing tables remain empty and the devices do not see each other. 
No messages are received by the other.
What am I missing with how this lib is supposed to be used ? I am not very familiar with FreeRtos incase this issue is related to that.
I would greatly appreciate if you could help me trouble shoot. Thank you very much.

COLVERTYETY, 2024-03-24:
I managed to get it working by modifying LM_SX1262.cpp:
```
LM_SX1262::LM_SX1262(uint8_t loraCs, uint8_t loraIrq, uint8_t loraRst, uint8_t loraIo1, SPIClass* spi) {
    // module = new SX1262(new Module(loraCs, loraIrq, loraRst, loraIo1, *spi));
    module = new SX1262(new Module(RADIO_CS_PIN, RADIO_DIO1_PIN, RADIO_RST_PIN, RADIO_BUSY_PIN));
}
```

Jaimi5, 2024-03-28:
Hi @COLVERTYETY!

Thanks for reaching out and using LoRaMesher!

It seams that you made a mistake assigning the PINS when initializing the code!

```    
    config.module = LoraMesher::LoraModules::SX1262_MOD;
    config.loraCs = RADIO_CS_PIN;
    config.loraIrq = RADIO_DIO0_PIN;
    config.loraRst = RADIO_RST_PIN;
    config.loraIo1 = RADIO_DIO1_PIN;
    config.freq = LoRa_frequency;
    //Init the loramesher with a processReceivedPackets function
    radio.begin(config);
```

The `config.loraIrq = RADIO_DIO0_PIN;` I think that should be the `config.loraIrq = RADIO_BUSY_PIN;` 

Can you confirm that this works?

Thank you very much!

COLVERTYETY, 2024-03-31:
Unfortunately changing the pin does not seem to solve the issue.
Thank you very much.

While I m here I have a few questions:
- sometimes I receive packets but the Routing Tables are not updated ? is this the expected behavior or what should I do to correct this.
- is there some kind of keep alive service/function/packet ?
- can you explain the gateway role ?

Great project, I like it a lot !

Jaimi5, 2024-04-07:
Hi @COLVERTYETY,
Could you try mirroring these settings for the pins you're configuring outside the function? This adjustment might do the trick:

```
    config.module = LoraMesher::LoraModules::SX1262_MOD;
    config.loraCs = RADIO_CS_PIN;
    config.loraIrq = RADIO_DIO1_PIN;
    config.loraRst = RADIO_RST_PIN;
    config.loraIo1 = RADIO_BUSY_PIN;
    config.freq = LoRa_frequency;
    //Init the loramesher with a processReceivedPackets function
    radio.begin(config);
```

For the questions you are asking:

> sometimes I receive packets but the Routing Tables are not updated ? is this the expected behavior or what should I do to correct this.

It's normal for routing table updates to be sent out periodically. If there aren't any changes, the system just keeps the connection alive without updating. This means your observation is expected under stable network conditions.

>is there some kind of keep alive service/function/packet ?

Yes, there's a keep alive mechanism in place. It's set to consider a device inactive if it misses 5 consecutive routing packets. This way, if a device doesn't show up in 5 routing updates, it's removed from the routing table, and this update cascades through the network. This can be changed through one of the `DEFAULT_TIMEOUT` in BuildOptions.h.

>can you explain the gateway role ?

The gateway serves as a bridge, just like any other role, but it's special because it can connect your mesh network to the internet or another network. You can dynamically assign or remove this role using addGatewayRole();, removeGatewayRole();, or RoleService::setRole(ROLE_GATEWAY); commands. The system ensures that this role's status is send to other nodes of the network using the Routing packets and, on each device, you can use `getClosestGateway()` or `getBestNodeWithRole` to get the address of the closest device with that role.

I hope this clears up your queries! If you've got more questions or need further assistance, feel free to drop a message.

Jaimi5, 2026-10-08:
Answered above. In 2.0.0 use `PinConfig` with `RadioType::kSx1262`. Closing.

closed 2026-10-08 by Jaimi5