LoRaMesher/LoRaMesher #44: Consulta sobre modulo SX1278

Pipoteex, 2024-04-20:
Hola buenas! Muy buen proyecto!

Te queria consultar por alguna idea o ayuda que me puedas brindar. Desde ya disculpa la molestias...

Resulta que tengo 3 modulos loras RA-02 1278 433mhz que se tienen que comunicar entre si para transmitir informacion. La idea es que ambos puedan recibir paquetes y enviarlos pero estos paquetes tienen que ser filtrados, por ejemplo: El modulo LORA 2 solo puede recibir paquetes de LORA 1 y no de LORA 3. 
Lo que pude ver es que los paquetes se pisan, a veces recibe y otra veces no.
La idea es que LORA 2 reciba paquetes de LORA 1 y LORA 3 pero que solo analice los que vienen de LORA 1.
No pude aun lograr eso.
Estuve probando otras librerias para esto por ejemplo RadioLib, LoRa y ahora pude leer un articulo sobre lu libreria.

Desde ya te agradezco mucho!

Jaimi5, 2024-05-01:
Buenos días @Pipoteex!

LoRaMesher te permitiría hacer esto y muchas cosas mas! No habría problema!

Si tienes alguna duda de como se usa la aplicación no dudes en contactarme.
Lo primero será configurar correctamente la radio, luego todo lo otro es muy senzillo.

Muchas gracias y suerte!

closed 2024-05-23 by Jaimi5

Pipoteex, 2024-05-25:
Hola buenos dias! No habia visto tu respuesta, desde ya agradezco tu tiempo.

Me gustaria saber como se configura la radio?

Saludos!

Jaimi5, 2024-05-26:
Buenas!

He actualizado el README y los ejemplos de LoRaMesher. Te recomiendo empezar con el ejemplo basico: [Codigo](https://github.com/LoRaMesher/LoRaMesher/tree/main/examples/Counter).

Si pones este mismo codigo en dos dispositivos deberias empezar a ver como llegan mensajes a cada uno de ellos.

Si tienes cualquier otra duda aqui estoy! :D

dliebel, 2024-07-13:
Hola que tal, muy buen proyecto... Les felicito

> Hola buenos dias! No habia visto tu respuesta, desde ya agradezco tu tiempo.
> 
> Me gustaria saber como se configura la radio?
> 
> Saludos!

Estaba tratando de configurar una LoRa32,  específicamente este modelo 

[image]

Cree un proyecto plaformio y lo configure así

```
[env:ttgo-lora32-v21]
platform = espressif32
board = ttgo-lora32-v21
framework = arduino
monitor_speed = 115200
monitor_filters = colorize

lib_deps =
	https://github.com/LoRaMesher/LoRaMesher.git

build_type = debug

;Use the following build_flags to enable debug output to the library
build_flags =
	-DCORE_DEBUG_LEVEL=5

```
Estoy usando el ejemplo counter de la carpeta de ejemplo,  use esto como configurarcion 

  //definiciones de lora
```
#define RADIO_SCLK_PIN              5
#define RADIO_MISO_PIN              19
#define RADIO_MOSI_PIN              27
#define RADIO_CS_PIN                18
#define RADIO_DIO0_PIN              26
#define RADIO_RST_PIN               23
#define RADIO_DIO1_PIN              33
#define RADIO_BUSY_PIN              32

#define CONFIG_RADIO_FREQ           433.0
#define CONFIG_RADIO_OUTPUT_POWER   17
#define CONFIG_RADIO_BW             125.0
```
y  en setupLoraMesher, tuve algunos inconvenientes con SX1278_MOD,  ademas   tuve que agregar estas librería
#include <WiFi.h> #include <esp_wifi.h>  por que se colgaban y reiniciaba. 

```
    //Get the configuration of the LoRaMesher
    LoraMesher::LoraMesherConfig config = LoraMesher::LoraMesherConfig();

     SPI.begin(RADIO_SCLK_PIN, RADIO_MISO_PIN, RADIO_MOSI_PIN);
    config.spi=&SPI;
    config.freq=433.0; //869.900F

    //configuracion t3 1.6.1
    config.loraCs = RADIO_CS_PIN;
    config.loraIrq = RADIO_DIO1_PIN;
    config.loraRst = RADIO_RST_PIN;
    config.loraIo1 = RADIO_BUSY_PIN;
    config.freq = CONFIG_RADIO_FREQ;
    config.power = CONFIG_RADIO_OUTPUT_POWER;
    config.bw = CONFIG_RADIO_BW;

    config.module = LoraMesher::LoraModules::SX1278_MOD;
[... 48 lines cut ...]
[  5844][I][RoutingTableService.cpp:216] manageTimeoutRoutingTable(): [LoRaMesher] Checking routes timeout
R  5861][V][LoraMesher.cpp:709] queueManager(): [LoRaMesher] No packets to send [o r  5r8e6c1e]i[vIe]d[
outingTableService.cpp:192] printRoutingTable(): [LoRaMesher] Current routing table:
Lora initialized
[  7799][V][LoraMesher.cpp:581] sendHelloPacket(): [LoRaMesher] Creating Routing Packet
[  7806][V][LoraMesher.cpp:582] sendHelloPacket(): [LoRaMesher] Stack space unused after entering the task: 2656
[  7817][V][LoraMesher.cpp:583] sendHelloPacket(): [LoRaMesher] Free heap: 311096
[  7824][V][PacketService.h:312] createPacket(): [LoRaMesher] Creating packet with 8 bytes
[  7832][I][PacketService.h:325] createPacket(): [LoRaMesher] Packet created with 8 bytes
[  7840][I][LoraMesher.h:649] setPackedForSend(): [LoRaMesher] Adding packet to Q_SP
[  7848][I][LoraMesher.h:651] setPackedForSend(): [LoRaMesher] Created packet to Q_SP
[  7855][I][LoraMesher.cpp:1010] addToSendOrderedAndNotify(): [LoRaMesher] Added packet to Q_SP, notifying sender task
[  7866][V][LoraMesher.cpp:494] sendPackets(): [LoRaMesher] Stack space unused after entering the task: 2528
[  7876][V][LoraMesher.cpp:495] sendPackets(): [LoRaMesher] Free heap: 311012
[  7883][V][LoraMesher.cpp:501] sendPackets(): [LoRaMesher] Size of Send Packets Queue: 1
[  7891][V][LoraMesher.cpp:508] sendPackets(): [LoRaMesher] Send n. 0
[  7897][V][LoraMesher.cpp:739] printHeaderPacket(): [LoRaMesher] Packet send -- Size: 8 Src: 74C Dst: FFFF Id: 0 Type: 4 Via: 0 Seq_Id: 0 Num: 0
[  7979][E][LoraMesher.cpp:465] sendPacket(): [LoRaMesher] Transmit gave error: -5
[  7988][V][LoraMesher.cpp:558] sendPackets(): [LoRaMesher] TimeOnAir 43 ms, next message in 0 ms
[  7996][E][LoraMesher.cpp:559] sendPackets(): [LoRaMesher] datos de 74C, destination FFFF
[  8004][I][PacketQueueService.h:80] deleteQueuePacketAndPacket(): [LoRaMesher] Deleting packet
[  8013][I][PacketQueueService.h:83] deleteQueuePacketAndPacket(): [LoRaMesher] Deleting packet queue
```

Muchas gracias de antemano