LoRaMesher/LoRaMesher #45: Reliable packets not working

PureTek-Innovations, 2024-05-04:
I am using an ESP32-S3 and SX1262.
There is an issue with reliable packets sometimes causing a corrupt heap, it seems to be coming from the "Queue Manager routine":

`V (640368) LoRaMesher: Free heap: 228288
I (640368) LoRaMesher: Packet created with 12 bytes
I (640379) LoRaMesher: Receiving LoRa packet: Size: 12 bytes RSSI: -30 SNR: 12
V (640381) LoRaMesher: Size of Received Packets Queue: 1
V (640391) LoRaMesher: Packet received -- Size: 12 Src: 1D78 Dst: BEEF Id: 14 Type: 83 Via: BEEF Seq_Id: 3 Num: 2
I (640402) LoRaMesher: Data packet from 1D78, destination BEEF, via BEEF
V (640412) LoRaMesher: Data packet from 1D78 for me
V (640412) LoRaMesher: Synchronization Packet received
V (640422) LoRaMesher: Starting to calculate RTT seq_Id: 3 Src: 1D78
V (640423) LoRaMesher: Timeout set to 59 s
V (640433) LoRaMesher: Creating packet with 12 bytes
V (640433) LoRaMesher: Stack space unused after entering the task: 1988
I (640433) LoRaMesher: Packet created with 12 bytes
V (640443) LoRaMesher: Free heap: 227960
I (640443) LoRaMesher: Adding packet to Q_SP
CORRUPT HEAP: Bad head at 0x3fcb2b64. Expected 0xabba1234 got 0x3fca20b4

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377d86:0x3fcadf30 0x4037d8c5:0x3fcadf50 0x403846d1:0x3fcadf70 0x403842e1:0x3fcae0a0 0x403785bd:0x3fcae0c0 0x40384701:0x3fcae0e0 0x4208c7a5:0x3fcae100 0x4208cd69:0x3fcae120 0x42098319:0x3fcae140 0x4209832c:0x3fcae160 0x4209869d:0x3fcae180 0x42098ebe:0x3fcae1b0 0x42098ed7:0x3fcae1d0 0x420107f5:0x3fcae1f0 0x42013dbf:0x3fcae210 0x42013e75:0x3fcae230 0x42003b59:0x3fcae250 0x420bc231:0x3fcae390 0x4203f326:0x3fcae3c0 0x4203f3bb:0x3fcae3f0 0x420076e5:0x3fcae470 0x4200784c:0x3fcae4d0 0x42007913:0x3fcae4f0 0x4200792d:0x3fcae510

  #0  0x40377d86:0x3fcadf30 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037d8c5:0x3fcadf50 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x403846d1:0x3fcadf70 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x403842e1:0x3fcae0a0 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:259 (discriminator 1)
  #4  0x403785bd:0x3fcae0c0 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:382
  #5  0x40384701:0x3fcae0e0 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x4208c7a5:0x3fcae100 in operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49        
  #7  0x4208cd69:0x3fcae120 in operator delete[](void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_opv.cc:35     
  #8  0x42098319:0x3fcae140 in AsyncWebSocketMessageBuffer::~AsyncWebSocketMessageBuffer() at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:264 (discriminator 1)
  #9  0x4209832c:0x3fcae160 in std::_Function_handler<void (AsyncWebSocketMessageBuffer* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketMessageBuffer*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1247
      (inlined by) _M_invoke at c:\users\jem\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:297
  #10 0x4209869d:0x3fcae180 in std::function<void (AsyncWebSocketMessageBuffer* const&)>::operator()(AsyncWebSocketMessageBuffer* const&) const at c:\users\jem\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::remove(AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/StringArray.h:258
      (inlined by) AsyncWebSocket::_cleanBuffers() at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1836
  #11 0x42098ebe:0x3fcae1b0 in AsyncWebSocket::textAll(AsyncWebSocketMessageBuffer*) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1415      
  #12 0x42098ed7:0x3fcae1d0 in AsyncWebSocket::textAll(char const*, unsigned int) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1423
  #13 0x420107f5:0x3fcae1f0 in WebSerialClass::write(unsigned char const*, unsigned int) at .pio/libdeps/esp32-s3-devkitc-1/WebSerial/src/WebSerial.cpp:52
  #14 0x42013dbf:0x3fcae210 in Print::write(char const*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:67
  #15 0x42013e75:0x3fcae230 in Print::print(char const*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:84
  #16 0x42003b59:0x3fcae250 in print_to_log_sd_card(char const*, __va_list_tag) at src/main.cpp:102
  #17 0x420bc231:0x3fcae390 in esp_log_writev at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/log/log.c:200
  #18 0x4203f326:0x3fcae3c0 in __wrap_esp_log_writev at /Users/ficeto/Desktop/ESP32/ESP32S2/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:452
  #19 0x4203f3bb:0x3fcae3f0 in __wrap_esp_log_write at /Users/ficeto/Desktop/ESP32/ESP32S2/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:469
  #20 0x420076e5:0x3fcae470 in LoraMesher::managerTimeouts(LM_LinkedList<LoraMesher::listConfiguration>*, LoraMesher::QueueType) at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:1408
  #21 0x4200784c:0x3fcae4d0 in LoraMesher::managerReceivedQueue() at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:1392
  #22 0x42007913:0x3fcae4f0 in LoraMesher::queueManager() at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:706 (discriminator 4)
  #23 0x4200792d:0x3fcae510 in LoraMesher::initializeSchedulers()::{lambda(void*)#6}::_FUN(void*) at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:305
      (inlined by) _FUN at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:305

ELF file SHA256: 39935a53cfa21356`

PureTek-Innovations, 2024-05-04:
A slightly different error:

`I (2658009) LoRaMesher: Packet created with 12 bytes
I (2658020) LoRaMesher: Receiving LoRa packet: Size: 12 bytes RSSI: -47 SNR: 13
V (2658022) LoRaMesher: Size of Received Packets Queue: 1
V (2658033) LoRaMesher: Packet received -- Size: 12 Src: 1D78 Dst: BEEF Id: 23 Type: 83 Via: BEEF Seq_Id: 73 Num: 2
I (2658043) LoRaMesher: Data packet from 1D78, destination BEEF, via BEEF
V (2658053) LoRaMesher: Data packet from 1D78 for me
V (2658053) LoRaMesher: Synchronization Packet received
V (2658064) LoRaMesher: Starting to calculate RTT seq_Id: 73 Src: 1D78
V (2658064) LoRaMesher: Timeout set to 65 s
V (2658074) LoRaMesher: Creating packet with 12 bytes
V (2658074) LoRaMesher: Stack space unused after entering the task: 1584
I (2658074) LoRaMesher: Packet created with 12 bytes
CORRUPT HEAP: Bad head at 0x3fcb2a20. Expected 0xabba1234 got 0x3fca20b4

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377d86:0x3fcabbe0 0x4037d8c5:0x3fcabc00 0x403846d1:0x3fcabc20 0x403842e1:0x3fcabd50 0x403785bd:0x3fcabd70 0x40384701:0x3fcabd90 0x4208be49:0x3fcabdb0 0x4208c40d:0x3fcabdd0 0x420979bd:0x3fcabdf0 0x420979d0:0x3fcabe10 0x42097d41:0x3fcabe30 0x42098562:0x3fcabe60 0x4209857b:0x3fcabe80 0x420100cd:0x3fcabea0 0x42013537:0x3fcabec0 0x420135ed:0x3fcabee0 0x42003bc9:0x3fcabf00 0x420bb8e1:0x3fcac040 0x4203ea9e:0x3fcac070 0x4203eb33:0x3fcac0a0 0x42003ca9:0x3fcac120 0x42005ebb:0x3fcac140 0x42006a6d:0x3fcac160 0x42006dd8:0x3fcac190 0x42006ea3:0x3fcac1b0 0x42007622:0x3fcac1e0 0x42007661:0x3fcac220

  #0  0x40377d86:0x3fcabbe0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037d8c5:0x3fcabc00 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x403846d1:0x3fcabc20 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x403842e1:0x3fcabd50 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:259 (discriminator 1)
  #4  0x403785bd:0x3fcabd70 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:382
  #5  0x40384701:0x3fcabd90 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x4208be49:0x3fcabdb0 in operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49        
  #7  0x4208c40d:0x3fcabdd0 in operator delete[](void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_opv.cc:35     
  #8  0x420979bd:0x3fcabdf0 in AsyncWebSocketMessageBuffer::~AsyncWebSocketMessageBuffer() at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:264 (discriminator 1)
  #9  0x420979d0:0x3fcabe10 in std::_Function_handler<void (AsyncWebSocketMessageBuffer* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketMessageBuffer*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1247
      (inlined by) _M_invoke at c:\users\jem\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:297
  #10 0x42097d41:0x3fcabe30 in std::function<void (AsyncWebSocketMessageBuffer* const&)>::operator()(AsyncWebSocketMessageBuffer* const&) const at c:\users\jem\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::remove(AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/StringArray.h:258
      (inlined by) AsyncWebSocket::_cleanBuffers() at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1836
  #11 0x42098562:0x3fcabe60 in AsyncWebSocket::textAll(AsyncWebSocketMessageBuffer*) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1415      
  #12 0x4209857b:0x3fcabe80 in AsyncWebSocket::textAll(char const*, unsigned int) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1423
  #13 0x420100cd:0x3fcabea0 in WebSerialClass::write(unsigned char const*, unsigned int) at .pio/libdeps/esp32-s3-devkitc-1/WebSerial/src/WebSerial.cpp:52
  #14 0x42013537:0x3fcabec0 in Print::write(char const*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:67
  #15 0x420135ed:0x3fcabee0 in Print::print(char const*) at C:/Users/Jem/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:84
  #16 0x42003bc9:0x3fcabf00 in print_to_log_sd_card(char const*, __va_list_tag) at src/main.cpp:105
  #17 0x420bb8e1:0x3fcac040 in esp_log_writev at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/log/log.c:200
  #18 0x4203ea9e:0x3fcac070 in __wrap_esp_log_writev at /Users/ficeto/Desktop/ESP32/ESP32S2/esp32-arduino-lib-builder/components/esp-rainmaker/components/esp-insights/components/esp_diagnostics/src/esp_diagnostics_log_hook.c:452
[... cut ...]
  #24 0x42006ea3:0x3fcac1b0 in LoraMesher::processDataPacket(QueuePacket<DataPacket>*) at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:841 (discriminator 7)
  #25 0x42007622:0x3fcac1e0 in LoraMesher::processPackets() at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:650
  #26 0x42007661:0x3fcac220 in LoraMesher::initializeSchedulers()::{lambda(void*)#4}::_FUN(void*) at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:285
      (inlined by) _FUN at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:285

ELF file SHA256: cefafbe3c162e770`

PureTek-Innovations, 2024-05-05:
This appears to just be a memory size issue.
I've doubled the allocated memory for each of the LoraMesher tasks and it now seems to be stable.
I'll confirm when the network has been up for a few days.

PureTek-Innovations, 2024-05-05:
Sadly it's still crashing, just not as often.
Here's the latest error message:

`V (6641055) LoRaMesher: No packets to send or received
V (6642184) LoRaMesher: Creating Routing Packet
V (6642184) LoRaMesher: Stack space unused after entering the task: 5928
V (6642184) LoRaMesher: Free heap: 207920
V (6642195) LoRaMesher: Creating packet with 12 bytes
I (6642195) LoRaMesher: Packet created with 12 bytes
I (6642205) LoRaMesher: Adding packet to Q_SP
I (6642205) LoRaMesher: Created packet to Q_SP
I (6642215) LoRaMesher: Added packet to Q_SP, notifying sender task
V (6642216) LoRaMesher: Stack space unused after entering the task: 5744
V (6642226) LoRaMesher: Free heap: 207816
V (6642226) LoRaMesher: Size of Send Packets Queue: 1
V (6642237) LoRaMesher: Send n. 293
V (6642237) LoRaMesher: RandomDelay 849 ms
V (6642469) LoRaMesher: Stack space unused after entering the task: 5864
V (6642470) LoRaMesher: Free heap: 207844
I (6642470) LoRaMesher: Packet created with 12 bytes
I (6642481) LoRaMesher: Receiving LoRa packet: Size: 12 bytes RSSI: -42 SNR: 12
V (6642483) LoRaMesher: Size of Received Packets Queue: 1
V (6642493) LoRaMesher: Packet received -- Size: 12 Src: 1D78 Dst: BEEF Id: 41 Type: 83 Via: BEEF Seq_Id: 84 Num: 2
I (6642504) LoRaMesher: Data packet from 1D78, destination BEEF, via BEEF
V (6642514) LoRaMesher: Data packet from 1D78 for me
V (6642514) LoRaMesher: Synchronization Packet received
V (6642525) LoRaMesher: Starting to calculate RTT seq_Id: 84 Src: 1D78
V (6642525) LoRaMesher: Timeout set to 65 s
V (6642535) LoRaMesher: Creating packet with 12 bytes
I (6642535) LoRaMesher: Packet created with 12 bytes
I (6642546) LoRaMesher: Adding packet to Q_SP
V (6642535) LoRaMesher: Stack space unused after entering the task: 5760
I (6642546) LoRaMesher: Created packet to Q_SP
I (6642556) LoRaMesher: Added packet to Q_SP, notifying sender task
I (6642567) LoRaMesher: Deleting packet
I (6642567) LoRaMesher: Deleting packet queue
V (6642557) LoRaMesher: Free heap: 207440
V (6642567) LoRaMesher: Stack space unused after entering the task: 5600
CORRUPT HEAP: Bad head at 0x3fcb7eac. Expected 0xabba1234 got 0x3fca20b4

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377d86:0x3fcb3100 0x4037d8c5:0x3fcb3120 0x403846d1:0x3fcb3140 0x403842e1:0x3fcb3270 0x403785bd:0x3fcb3290 0x40384701:0x3fcb32b0 0x4208c719:0x3fcb32d0 0x4208ccdd:0x3fcb32f0 0x4209828d:0x3fcb3310 0x420982a0:0x3fcb3330 0x42098611:0x3fcb3350 0x42098e32:0x3fcb3380 0x42098e4b:0x3fcb33a0 0x420107b9:0x3fcb33c0 0x42013d83:0x3fcb33e0 0x42013e39:0x3fcb3400 0x42003ba1:0x3fcb3420 0x420bc1b1:0x3fcb3560 0x4203f2ea:0x3fcb3590 0x4203f37f:0x3fcb35c0 0x420076a9:0x3fcb3640 0x42007810:0x3fcb36a0 0x420078d7:0x3fcb36c0 0x420078f1:0x3fcb36e0

  #0  0x40377d86:0x3fcb3100 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037d8c5:0x3fcb3120 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x403846d1:0x3fcb3140 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x403842e1:0x3fcb3270 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:259 (discriminator 1)
  #4  0x403785bd:0x3fcb3290 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:382
  #5  0x40384701:0x3fcb32b0 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x4208c719:0x3fcb32d0 in operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49        
  #7  0x4208ccdd:0x3fcb32f0 in operator delete[](void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_opv.cc:35     
  #8  0x4209828d:0x3fcb3310 in AsyncWebSocketMessageBuffer::~AsyncWebSocketMessageBuffer() at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:264 (discriminator 1)
  #9  0x420982a0:0x3fcb3330 in std::_Function_handler<void (AsyncWebSocketMessageBuffer* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketMessageBuffer*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1247
      (inlined by) _M_invoke at c:\users\jem\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:297
  #10 0x42098611:0x3fcb3350 in std::function<void (AsyncWebSocketMessageBuffer* const&)>::operator()(AsyncWebSocketMessageBuffer* const&) const at c:\users\jem\.platformio\packages\toolchain-xtensa-esp32s3\xtensa-esp32s3-elf\include\c++\8.4.0\bits/std_function.h:687
      (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::remove(AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/StringArray.h:258
      (inlined by) AsyncWebSocket::_cleanBuffers() at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1836
  #11 0x42098e32:0x3fcb3380 in AsyncWebSocket::textAll(AsyncWebSocketMessageBuffer*) at .pio/libdeps/esp32-s3-devkitc-1/AsyncWebServer_ESP32_SC_W5500/src/AsyncWebSocket.cpp:1415      
[... cut ...]
  #21 0x42007810:0x3fcb36a0 in LoraMesher::managerReceivedQueue() at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:1392
  #22 0x420078d7:0x3fcb36c0 in LoraMesher::queueManager() at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:706 (discriminator 4)
  #23 0x420078f1:0x3fcb36e0 in LoraMesher::initializeSchedulers()::{lambda(void*)#6}::_FUN(void*) at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:305
      (inlined by) _FUN at .pio/libdeps/esp32-s3-devkitc-1/LoRaMesher/src/LoraMesher.cpp:305

ELF file SHA256: 7a6df068d8ca6ddb`

Jaimi5, 2026-10-08:
Legacy 0.0.x code, replaced in 2.0.0. Closing. Please open a new issue if it happens on 2.0.0.

closed 2026-10-08 by Jaimi5