LoRaMesher/LoRaMesher #49: Corrupted routing table

PureTek-Innovations, 2024-05-06:
I left a test network of ESP32-S3 SX1262s running overnight.
The routing table has become corrupted:

I (9750303) LoRaMesher: Current routing table:
I (9750303) LoRaMesher: 0 - 1CC4 via 1CC4 metric 1 Role 0
I (9750303) LoRaMesher: 1 - 2024 via 2024 metric 1 Role 0
I (9750313) LoRaMesher: 2 - 212C via 212C metric 1 Role 0
I (9750313) LoRaMesher: 3 - 1D78 via 1D78 metric 1 Role 0
I (9750323) LoRaMesher: 4 - 800 via 2024 metric 3 Role 120
I (9750323) LoRaMesher: 5 - A800 via 212C metric 4 Role 0
I (9750334) LoRaMesher: 6 - 54 via 1D78 metric 4 Role 0
I (9750334) LoRaMesher: 7 - A7BE via 212C metric 5 Role 0
I (9750344) LoRaMesher: 8 - 0 via 212C metric 4 Role 0
I (9750344) LoRaMesher: 9 - 60 via 212C metric 3 Role 0
I (9750354) LoRaMesher: 10 - FFFF via 212C metric 4 Role 255
I (9750354) LoRaMesher: Deleting packet
I (9750365) LoRaMesher: Deleting packet queue

Nodes 0 to 3 are genuine, the rest don't exist.
The on the gateway has been increasing since 03:45 and is currently about 35, which seems very high.
[image]

The gateway is no longer receiving/processing data packets from two nodes and only some packets are being processed for the other two.

I'll attach the log files from the gateway, the test network nodes don't have SD cards yet.

PureTek-Innovations, 2024-05-06:
I should add that the gateway appears to have rebooted just before 07:00

PureTek-Innovations, 2024-05-06:
[LOG00030.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218662/LOG00030.TXT)
[LOG00031.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218663/LOG00031.TXT)
[LOG00032.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218664/LOG00032.TXT)
[LOG00033.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218665/LOG00033.TXT)
[LOG00034.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218666/LOG00034.TXT)
[LOG00035_1.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218681/LOG00035_1.TXT)
[LOG00035_2.TXT](https://github.com/LoRaMesher/LoRaMesher/files/15218682/LOG00035_2.TXT)

Jaimi5, 2026-10-08:
Legacy 0.0.x code, replaced in 2.0.0. Closing. Please open a new issue if it happens on 2.0.0.

closed 2026-10-08 by Jaimi5