LoRaMesher/LoRaMesher #50: Gateway deleting data packets after reboot

PureTek-Innovations, 2024-05-06:
I almost always lose a data packet or two after the gateway reboots:

I (78895) LoRaMesher: Packet created with 12 bytes
I (78906) LoRaMesher: Receiving LoRa packet: Size: 12 bytes RSSI: -78 SNR: 8
V (78908) LoRaMesher: Size of Received Packets Queue: 1
V (78918) LoRaMesher: Packet received -- Size: 12 Src: 1D78 Dst: BEEF Id: 1 Type: 83 Via: BEEF Seq_Id: 0 Num: 2
I (78928) LoRaMesher: Data packet from 1D78, destination BEEF, via BEEF
V (78939) LoRaMesher: Data packet from 1D78 for me
V (78939) LoRaMesher: Synchronization Packet received
W (78949) LoRaMesher: Node not found in the routing table
I (78949) LoRaMesher: Deleting packet
I (78959) LoRaMesher: Deleting packet queue

If a node receives a correctly formatted sync packet or a data packet from a node that is not in the routing table, it seems to me that it would be better to add that node to the routing table rather than to delete the packet.

Jaimi5, 2024-05-23:
This is usefull since sometimes you receive some packet with garbage and could be forwarded to the application. This prevents some of this packets.
This can be removed easily if you want.

Jaimi5, 2026-10-08:
Legacy 0.0.x code, replaced in 2.0.0. Closing. Please open a new issue if it happens on 2.0.0.

closed 2026-10-08 by Jaimi5