LoRaMesher/LoRaMesher #51: Nodes not forwarding reliable packets

PureTek-Innovations, 2024-05-06:
I need to do more investigation on this, ideally adding the SD cards to the nodes.

When I set my test network up such that one of the nodes cannot communicate directly with the gateway, none of its reliable data packets get to the gateway.  Here's the routing table:

I (1491794) LoRaMesher: Current routing table:
I (1491804) LoRaMesher: 0 - 1D78 via 1D78 metric 1 Role 0
I (1491804) LoRaMesher: 1 - 2024 via 1D78 metric 2 Role 0
I (1491814) LoRaMesher: 2 - 1CC4 via 1CC4 metric 1 Role 0
I (1491814) LoRaMesher: 3 - 212C via 212C metric 1 Role 0

The gateway receives no reliable packets from node 2024

PureTek-Innovations, 2024-05-06:
Just saw this in the gateway log.  The gateway is BEEF and the problem node is 2024

13:30:04.341 -> I (1901066) LoRaMesher: Packet created with 12 bytes
13:30:04.380 -> I (1901067) LoRaMesher: Receiving LoRa packet: Size: 12 bytes RSSI: -74 SNR: 8 FREQ ERR: 0.000000
13:30:04.380 -> V (1901084) LoRaMesher: Size of Received Packets Queue: 1
13:30:04.380 -> V (1901084) LoRaMesher: Packet received -- Size: 12 Src: BEEF Dst: 2024 Id: 167 Type: 34 Via: 2024 Seq_Id: 2 Num: 1
13:30:04.380 -> I (1901094) LoRaMesher: Data packet from BEEF, destination 2024, via 2024
13:30:04.425 -> V (1901105) LoRaMesher: Packet not for me, deleting it

Jaimi5, 2026-10-08:
Legacy 0.0.x code, replaced in 2.0.0. Closing. Please open a new issue if it happens on 2.0.0.

closed 2026-10-08 by Jaimi5