LoRaMesher/LoRaMesher #52: Request: port to nonArduino

onyx99, 2024-05-17:
Hello,

is it possible to use or port LoRaMesher to nonArduino?

RadioLib works fine here under Linux thanks to a Hal driver, see for example
https://github.com/jgromes/RadioLib/tree/master/examples/NonArduino/Raspberry

LoRaMesher seems to be based on RadioLib but with Mesh implemented it unfortunate seems to have lost compatibility for non arduino systems. (at least if i look to the code).

would really love to see LoRaMesher working under Linux,

Kind regards.

Jaimi5, 2024-05-23:
Hi!

The new version of LoRaMesher should be able to work without the Arduino.
Did you try to execute it without the Arduino?

Thanks!

onyx99, 2024-06-11:
Hi,

Yes i tried, i get errors like:

In file included from /mnt/project/LoRaMesher/src/BuildOptions.cpp:1:
/mnt/project/LoRaMesher/src/BuildOptions.h:7:10: fatal error: freertos/FreeRTOS.h: No such file or directory
    7 | #include <freertos/FreeRTOS.h>

In file included from /mnt/project/LoRaMesher/src/BuildOptions.cpp:1:
/mnt/project/LoRaMesher/src/BuildOptions.h:12:10: fatal error: esp_log.h: No such file or directory
   12 | #include <esp_log.h>

In file included from /mnt/project/LoRaMesher/src/BuildOptions.cpp:1:
/mnt/project/LoRaMesher/src/BuildOptions.h:13:10: fatal error: esp_heap_caps.h: No such file or directory
   13 | #include <esp_heap_caps.h>

/mnt/project/LoRaMesher/src/BuildOptions.cpp:14:10: fatal error: esp_timer.h: No such file or directory
   14 | #include <esp_timer.h>
      |          ^~~~~~~~~~~~~

/mnt/project/LoRaMesher/src/BuildOptions.cpp: In function ‘size_t getFreeHeap()’:
/mnt/project/LoRaMesher/src/BuildOptions.cpp:17:36: error: ‘MALLOC_CAP_INTERNAL’ was not declared in this scope
   17 |     return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
      |                                    ^~~~~~~~~~~~~~~~~~~
/mnt/project/LoRaMesher/src/BuildOptions.cpp:17:12: error: ‘heap_caps_get_free_size’ was not declared in this scope
   17 |     return heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
      |            ^~~~~~~~~~~~~~~~~~~~~~~
/mnt/project/LoRaMesher/src/BuildOptions.cpp: At global scope:
/mnt/project/LoRaMesher/src/BuildOptions.cpp:20:25: error: expected initializer before ‘millis’
   20 | unsigned long IRAM_ATTR millis() {
      |                         ^~~~~~

ToshikSoni, 2025-06-20:
Faced the same errors while trying to run on Raspberry Pi 4. is there any workaround for this? LoRaMesher is quite dependent on ESP and Arduino libraries. I am trying to include as many dependencies as I can but the list keeps on increasing. I will update here if I am able to find a solution.