LoRaMesher/LoRaMesher #57: Possible with Heltec Wireless Tracker?

karlscholz, 2024-08-15:
Hi, i would love to use your project with the Heltec Wireless Tracker, but I am facing some issues initializing the LoRa Module with Error -2, which as I looked into it means that it couldn't find it. As it has a SX1262 onboard, I am using the SX1262 example project and got the pin numbering from the official heltec library, which works. Then i thought i could just change the pin numbering in
https://github.com/LoRaMesher/LoRaMesher/blob/main/examples/SX1262/src/main.cpp#L113
before it gets initialized by setting it up like this:
```
void setupLoraMesher()
{
    // Example on how to change the module. See LoraMesherConfig to see all the configurable parameters.
    LoraMesher::LoraMesherConfig config;
    config.module = LoraMesher::LoraModules::SX1262_MOD;

    config.loraCs = 8;    // RADIO_NSS
    config.loraIrq = 13;  // RADIO_BUSY
    config.loraRst = 12;  // RADIO_RESET
    config.loraIo1 = 14;  // RADIO_DIO_1

    // Init the loramesher with a processReceivedPackets function
    radio.begin(config);
    ...
```

Trying to give it a new SPI configuration with
```
spi->begin(9,   // SCK
           11,  // MISO
           10); // MOSI
```

messed things up even further but i don't think it is needed due to the HAL.

Do you have any ideas or hints on how i could resolve my issue?

Jaimi5, 2024-08-15:
Hi!

Have you tried adding the spi to the config.spi like this?

```
SPIClass newSPI(HSPI);

void setupLoraMesher() {

    // Example on how to change the module. See LoraMesherConfig to see all the configurable parameters.
    LoraMesher::LoraMesherConfig config;
    config.module = LoraMesher::LoraModules::SX1262_MOD;

    config.loraCs = 8;    // RADIO_NSS
    config.loraIrq = 13;  // RADIO_BUSY
    config.loraRst = 12;  // RADIO_RESET
    config.loraIo1 = 14;  // RADIO_DIO_1

    newSPI.begin(9, 11, 10, 8); // Last is RADIO_NSS/CS
    config.spi = &newSPI;

    // Init the loramesher with a processReceivedPackets function
    radio.begin(config);
}

```

I think the problem is that you need to define the NSS/CS where the spi is initialized.
Let me know if this helps!

karlscholz, 2024-08-15:
Hi, thank you for the quick response. No it doesn't. regardless of if i put the slaveselect within the spi begin, changing the spi entry of the config. I was going to past you the error code here, it said something about GPIO and interrupt that something is not set up, but I can't recreate it since it just kinda bricked my ESP, i tried resetting it via 
```
python -m esptool --chip esp32s3 --port COM6 erase_flash
```
but it doesn't work anymore...

karlscholz, 2024-08-16:
I got the board working again and this is the output when i try to change anything with the config.spi:
```
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4200724c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (8058) gpio: gpio_isr_handler_remove(482): GPIO isr service is not installed, call gpio_install_isr_service() first
[  8179][E][LoraMesher.cpp:247] startReceiving(): [LoRaMesher] Starting receiving gave error: -2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4037ba45  PS      : 0x00060830  A0      : 0x82002416  A1      : 0x3fcee790  
A2      : 0x00000008  A3      : 0xffffffff  A4      : 0x00000008  A5      : 0x00000000  
A6      : 0x4201bb64  A7      : 0xff000000  A8      : 0x80380c78  A9      : 0x3fcee7a0  
A10     : 0x3fcec674  A11     : 0x3fcec674  A12     : 0x2d800000  A13     : 0x1f460000  
A14     : 0x02cf1d70  A15     : 0x00ffffff  SAR     : 0x00000012  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000048  LBEG    : 0x40056f08  LEND    : 0x40056f12  LCOUNT  : 0x00000000  

Backtrace: 0x4037ba42:0x3fcee790 0x42002413:0x3fcee7d0 0x4202121d:0x3fcee800 0x4201ae7d:0x3fcee820 0x4201b427:0x3fcee850 0x4201b494:0x3fcee890 0x4201bb5a:0x3fcee8d0 0x4201bb71:0x3fcee900 0x4201b966:0x3fcee920 0x42030cd7:0x3fcee950 0x42002c49:0x3fcee970 0x420041b2:0x3fcee9c0 0x42004235:0x3fceea00

ELF file SHA256: f75252c9ce407a0f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4200724c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (8058) gpio: gpio_isr_handler_remove(482): GPIO isr service is not installed, call gpio_install_isr_service() first
[  8179][E][LoraMesher.cpp:247] startReceiving(): [LoRaMesher] Starting receiving gave error: -2
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

Jaimi5, 2024-08-16:
The line you are seeing `E (8058) gpio: gpio_isr_handler_remove(482): GPIO isr service is not installed, call gpio_install_isr_service() first` is something that I have been seeing for a long time and I should not worry about it.

I have been doing some research and I have find that the RadioLib example for [SX1262](https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_PingPong/SX126x_PingPong.ino)  
Swaps the order of bussy and dio1. Could you try it?
```
// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1262 radio = new Module(10, 2, 3, 9);
```

karlscholz, 2024-08-16:
Ah okay, but it seemes that it causes the board to reboot and crash forever in a loop for me.

Interesting, i looked into the manufacturers [library](https://github.com/HelTecAutomation/Heltec_ESP32/blob/master/src/driver/board-config.h#L44-L53) and found this configuration. Also in this configuration the 10 as NSS would be the sames the supposed MOSI at also PIN 10. I tried it with and without changing the config.spi and it always gave this:
```
--- Terminal on COM6 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[ 13236][E][LoraMesher.cpp:215] initializeLoRa(): [LoRaMesher] Radio module gave error: -2
E (18126) gpio: gpio_isr_handler_remove(482): GPIO isr service is not installed, call gpio_install_isr_service() first
[ 18246][E][LoraMesher.cpp:247] startReceiving(): [LoRaMesher] Starting receiving gave error: -2
[ 20244][E][LoraMesher.cpp:247] startReceiving(): [LoRaMesher] Starting receiving gave error: -2
[ 29375][E][LoraMesher.cpp:215] initializeLoRa(): [LoRaMesher] Radio module gave error: -2
[ 29383][E][LoraMesher.cpp:247] startReceiving(): [LoRaMesher] Starting receiving gave error: -2
[ 31364][E][LoraMesher.cpp:215] initializeLoRa(): [LoRaMesher] Radio module gave error: -2
[ 31372][E][LoraMesher.cpp:247] startReceiving(): [LoRaMesher] Starting receiving gave error: -2
```

which means that it can't connect to the Radio right?

Jaimi5, 2026-10-08:
Legacy API. In 2.0.0 use `PinConfig` with `RadioType::kSx1262`. Closing.

closed 2026-10-08 by Jaimi5