LoRaMesher/LoRaMesher #62: Question about LM_LinkedList<T>::prev()

Bilel-Eljaamii, 2024-10-24:
Hi,

I have a question about the prev() function in LinkeeQueue.hpp file

``` c++
template<class T>
bool LM_LinkedList<T>::prev() {
    if (length == 0)
        return false;
    if (curr->prev != nullptr)
        return false;
    curr = curr->prev;
    return true;
}
```

As per my understanding this function will move the current iterator to the previous element in the queue.

1. The First if condition will check if the tje list is already empty, so no previous element 
2. second if: will check if the prev pointer if it is null or not
3. then the function will return true only if length >0 and  curr->prev **==** NULLptr . and curr->prev will get **always** _NULLptr_

I think this is a not the correct way to implement prev function.

- The issue is in the second IF condition, curr->prev shall be checked **if it is equal to nullptr ( == )** not **if it is deifferent (!=)**.
- if it is NULLptr means we are already in the head of the Queue and w can not get prev element.
- After we checked that The Queueu is not empty and we are not pointing on the first element we can set  curr = curr->prev; and return TRUE; 

This is my proposition how it shall be implemented :
``` c++
/**
 * @brief Move the current node to the previous node in the linked list.
 *
 * @return True if the operation was successful, false otherwise.
 */
template <class T>
bool LM_LinkedList<T>::prev()
{
    // Variable to track the state of the operation
    bool retVal = false;

    // Check if the list is not empty, the current node is not null and the previous node exists
    if ((length != 0) && (curr != nullptr) && (curr->prev != nullptr))
    {
        // Move to the previous node
        curr = curr->prev;
        retVal = true;
    }
    else
    {
        // 6.4.2: The 'if' ... 'else if' construct should be terminated with an 'else' statement.
    }
    return retVal;
}
```

Bilel-Eljaamii, 2024-10-24:
The function prev() is never called in the project, that's why it is a hidden bug.

Jaimi5, 2026-10-08:
`LM_LinkedList` was removed in the 1.0 rewrite. Closing.

closed 2026-10-08 by Jaimi5